fossasia/pslab-app #585: Rename diode_setup xml file

CloudyPadmal, 2017-08-25:
**Actual Behaviour**

The file has been used in many experiments as it shares the same behavior

**Expected Behaviour**

Since the other experiments using this layout file has no relevance to diode experiments, the file should bear a common name

**Steps to reproduce it**

> Not available

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Yes

asitava1998, 2017-08-25:
@CloudyPadmal Please send the PR for this after the existing PRs involving the `diode_setup.xml` file are merged. Otherwise the merge conflicts caused by this would be difficult to solve.

CloudyPadmal, 2017-08-25:
Sure @asitava1998 ! :+1: :100:

CloudyPadmal, 2017-08-26:
Fixes in [#589](https://github.com/fossasia/pslab-android/pull/589)

closed 2017-08-26 by CloudyPadmal