fossasia/pslab-app #586: demonstrate experiments with PSLab hardware device

viveksb007, 2017-08-25:
**Actual Behaviour**

Experiments are implemented, but there are no demos on how we should perform the experiment. There are experiment docs with schematics and basic theory.

**Expected Behaviour**

There should be some video demonstration of experiments. List of experiments for which demo should be there
- [ ]
- [ ]
- [ ] 

**Would you like to work on the issue?**

Let us know if this issue should be assigned to you or tell us who you think could help to solve this issue.

asitava1998, 2017-08-25:
About the written part, I think we should make a common template and write the experiment description and procedure according to it.

mariobehling, 2018-06-21:
Moving this here https://github.com/fossasia/pslab-experiments/issues/24

closed 2018-06-21 by mariobehling