fossasia/pslab-app #613: Implement Curve fitting in Oscilloscope

akarshan96, 2017-09-02:
**Actual Behaviour**

Curve fitting is not implemented in Oscilloscope yet.

**Expected Behaviour**

Curve fitting should be implemented in Oscilloscope yet.

**Steps to reproduce it**

Applications -> Oscilloscope

**LogCat for the issue**

None

**Screenshots of the issue**

None

**Would you like to work on the issue?**

Yes

Vikum94, 2018-05-04:
@akarshan96 I would like to work on this issue. Are you still working on it?

CloudyPadmal, 2018-05-04:
@Vikum94 you can continue!

akarshan96, 2018-05-05:
@Vikum94 let me give you a brief about what I had done but wasn't successful. For Sine Fit function in [AnalyticsClass.java](https://github.com/fossasia/pslab-android/blob/development/app/src/main/java/org/fossasia/pslab/communication/AnalyticsClass.java#L120) needs to be used. This function will return Amplitude, Frequency, Offset and Phase, these parameters then put into **Amplitude * sin(Frequency * time + Phase) + Offset** . But when I was trying it with the dataset which I received from the PSLab device itself I was getting inaccurate results. You give it a try too, let me also look into this issue again.
Maybe [this](https://github.com/akarshan96/pslab-android/commit/b70b61b3d88aaf480ebfb0e7ca3389512b8cb803#diff-000cee686dfe0defeb32ee0728dfbe95R503) will help you 
Cheers.

CloudyPadmal, 2018-05-05:
Thank you @akarshan96 for the heads up! :)

Vikum94, 2018-05-05:
Thank you very much! @akarshan96 Will see what I can do.

PR #1835 (2019-07-03): Curve fit in oscilloscope
Fixes #613

**Changes**: curve fitting working fine in oscilloscope

**Screenshot/s for the changes**: 
[image]

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra [... cut]
closed 2019-07-03 by CloudyPadmal