fossasia/pslab-app #630: No action bar in "About Us" activity

NamanRawat1, 2017-09-16:
**Actual Behaviour**
No action bar in About Us activity.
**Expected Behaviour**
There must be action bar in this activity like others activity with label About Us.

**Steps to reproduce it**
1.Open navigation Bar
2.Go to about us.

**Screenshots of the issue*
[image]

**Would you like to work on the issue?**
YES.

NamanRawat1, 2017-09-16:
Can I work on this?

CloudyPadmal, 2017-09-16:
Sure!! You may!! :)

viveksb007, 2017-09-16:
I don't think it needs an `ActionBar` with `About Us` written on it. Its pretty obvious when user clicked from `Nav Drawer`. Plus App don't have `ActionBar` in `About Us` Page.

NamanRawat1, 2017-09-16:
ok and also there is no navigation bar in this activity due to which user has to press back button to reach other activity.Is there a need for navigation view in this activity?

viveksb007, 2017-09-16:
@NamanRawat1 its an Activity. It can't have incorporate NavigationView of the MainActivity unless you replicate complete view of Main Activity. So I think we can go with `ActionBar` approach to take user to `MainActivity`. Please add an ActionBar with title `About PSLab`.
Thanks.

NamanRawat1, 2017-09-23:
All views of about us activity is from a string file. When I create a view for the toolbar in about us activity then two child element will clash. Is there any possible way to fix this problem? @viveksb007

viveksb007, 2017-09-23:
One way can be that you create a Root Layout(LinearLayout) with (`Toolbar` and `LinearLayout`) as children and add `aboutPage` view in the LinearLayout.
Hope the solution is clear.

viveksb007, 2017-09-28:
@NamanRawat1 Any updates on solving the issue? Should I send a PR if its not solved by you yet?

NamanRawat1, 2017-09-28:
just send pull request for it review it @viveksb007

viveksb007, 2017-10-02:
Fixed in #646

closed 2017-10-02 by viveksb007