fossasia/pslab-app #631: Implement Measure Voltages Experiment

CloudyPadmal, 2017-09-16:
**Actual Behaviour**

`Measure Voltages` experiment under saved experiments hasn't been implemented yet

**Expected Behaviour**

The experiment needs to be implemented

**Steps to reproduce it**

1. Open app
2. Go to `Saved Experiments`
3. Browse to `School Level` and try `Measure Voltages` experiment

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

If a new developer wants to take on this issue, can proceed with it! Otherwise I'll resolve this! :)

NamanRawat1, 2017-09-16:
Will I make a try on this?

CloudyPadmal, 2017-09-16:
Yea sure! :D

NamanRawat1, 2017-09-17:
[image]

How can i resolve this?

asitava1998, 2017-09-17:
@NamanRawat1  The doc is in markdown format. So, you need to use the `ExperimentDocMdFragment` so that this file is rendered by MarkdownView

NamanRawat1, 2017-09-17:
I already return ExperimentDocMdFragment in PerformExperimentAdapter.java. But it's not working.Can I post the code of this java file in the comment?

asitava1998, 2017-09-17:
@NamanRawat1  Push the code in your fork of this repository and give us the link of the same. We can suggest the changes there.

NamanRawat1, 2017-09-18:
https://github.com/NamanRawat1/pslab-android/commit/8964f0a0ef115b249706e5d5a2933830abc5d9fc

NamanRawat1, 2017-09-18:
[image]

Its working now can I send a pull request

asitava1998, 2017-09-18:
@NamanRawat1 Sure. Go ahead 👍

asitava1998, 2017-09-24:
Fixed in PR #633

closed 2017-09-24 by asitava1998