fossasia/pslab-app #632: Implement Resistance Measurement Experiment

CloudyPadmal, 2017-09-16:
**Actual Behaviour**

`Resistance Measurement` experiment under saved experiments hasn't been implemented yet

**Expected Behaviour**

The experiment needs to be implemented

**Steps to reproduce it**

1. Open app
2. Go to `Saved Experiments`
3. Browse to `School Level` and try `Resistance Measurement` experiment

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

If a new developer wants to take on this issue, can proceed with it! Otherwise I'll resolve this! :)

avish12, 2017-09-16:
shall I implement it?

CloudyPadmal, 2017-09-16:
Definitely! :)

avish12, 2017-09-16:
where can I find Experiment docs and Setup details?

CloudyPadmal, 2017-09-16:
Hi @avish12 ! You can find the resources from following links;
- [Python Implementation](https://github.com/fossasia/pslab-desktop-apps/tree/development/psl_res/GUI)
- [Documentation](https://github.com/fossasia/pslab-desktop-apps/tree/development/docs/_apps)
- [Android folders](https://github.com/fossasia/pslab-android/tree/development/app/src/main/java/org/fossasia/pslab/experimentsetup)

Hope that helps! :)

NamanRawat1, 2017-09-16:
Where is Measuring voltage experiment details in these document? I don't find it in documentation.

asitava1998, 2017-09-17:
@avish12 You can also have a look at the older PRs for experiments. They would give you the best idea

NamanRawat1, 2017-09-23:
@avish12  Are you working on this?

shikhart98, 2017-09-24:
Can I work on it?

asitava1998, 2017-09-24:
@avish12 @NamanRawat1 @shikhart98  Please resolve among yourselves who wants to work on this. Next time onwards if someone wants to work on an issue, assign yourself to that issue. The `assignees` section is at the top right. If the issue has been already assigned to someone, no one else can claim it.

NamanRawat1, 2017-09-24:
The issue is first given to @avish12  if he will not work on it then I will take it. Is that fine? @shikhart98

shikhart98, 2017-09-25:
Yea Sure!

avish12, 2017-09-25:
@shikhart98 if you want, you can work on it.

NamanRawat1, 2017-09-25:
@shikhart98 I already worked on the same type of issue you can work on it.

shikhart98, 2017-09-26:
Okay thanks!

shikhart98, 2017-09-29:
Sir can you please guide me with this issue. I'm a noob and new to open
source. I know JAVA and data structures/algos for now!

NamanRawat1, 2017-10-27:
Can I work on this issue if you are not working on it?  @shikhart98

sagarbaba, 2018-03-01:
I am working on this.

harsh-2711, 2018-03-01:
I have made it.. soon starting a PR

harsh-2711, 2018-03-01:
Also few others are completed by me. Will file PR by tomorrow

sagarbaba, 2018-03-01:
Leave this one for me.

harsh-2711, 2018-03-01:
Have u made the experiment for resistance measurement in python @sagarbaba

sagarbaba, 2018-03-01:
its already there in desktop app,

harsh-2711, 2018-03-01:
I mean have you implemented it in your PR. Or you have only made only the experiment setup tab

harsh-2711, 2018-03-01:
I have made both and is completely ready. I am getting it reviewed

CloudyPadmal, 2018-05-28:
Issues related to experiments will be considered once the Android app is functional

closed 2018-05-28 by CloudyPadmal