fossasia/pslab-app #634: reformat experiment heading and sub-heading in md experiment docs

viveksb007, 2017-09-19:
**Actual Behaviour**

Currently Heading, Sub-heading, titles all are of same type.  

**Expected Behaviour**

There should be difference Font size of Title, Heading, Sub-heading, etc.  

**Steps to reproduce it**

Any experiment doc of MD files.

**LogCat for the issue**

NA

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

For new devs.

NamanRawat1, 2017-09-23:
Can I work on this?

viveksb007, 2017-09-23:
Sure go ahead :+1:

viveksb007, 2017-09-28:
@NamanRawat1 any updates on the issue?

NamanRawat1, 2017-09-29:
when changing headings from h2 to h1 font size in main.css no change occurs.Trying to fix it.

viveksb007, 2017-09-29:
Now `md` files are rendered, so make change in `md` files, it would reflect in webView. We are actually shifting `html` files to `md` files, once thats completed, all `html` files would be removed.

commit 9b68aa0b (2017-10-09): Reformated experiments headings and subheadings (#651)

Fixes #634
closed 2017-10-09 by viveksb007