fossasia/pslab-app #641: Report issues website opens in browser

nikit19, 2017-09-28:
**Actual Behaviour**

When we click on report issues it opens up in browser which is slow.

**Expected Behaviour**

We should use custom tabs to open the website so that it happens quickly.

**Steps to reproduce it**

Open the app and click on Report issues from the drawer.

**Would you like to work on the issue?**

Yes!

CloudyPadmal, 2017-09-28:
Hi @nikit19 ! Can you elaborate more on the fix you're suggesting?

nikit19, 2017-09-28:
Hi @CloudyPadmal  If we use Chrome custom tabs the website could open much  faster. What do you think about it? https://developer.chrome.com/multidevice/android/customtabs

nikit19, 2017-09-28:
@CloudyPadmal It would look like this
[image]

CloudyPadmal, 2017-09-28:
Great! :D You can send a PR when done! :+1:

commit b81ee4ae (2017-09-29): Fixes #641 implement custom tab service (#642)
closed 2017-09-29 by viveksb007