fossasia/pslab-app #645: The logo is not visible/clear  with dark themed phones

Arshilgenius, 2017-09-28:
**Actual Behaviour**

The logo is not visible  because it's completely black and it should be changed to some other colour.

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

viveksb007, 2017-09-28:
@Arshilgenius Please refer Issue https://github.com/fossasia/pslab-artwork/issues/1

Arshilgenius, 2017-09-28:
@viveksb007  okay, should i close this issue?

viveksb007, 2017-09-28:
Nope you can leave it open. Once you create an Icon. Please post on our [gitter](https://gitter.im/fossasia/pslab) channel for review by community. And then you can send a PR here.

closed 2017-11-13 by Arshilgenius