fossasia/pslab-app #648: nav_header_main is not centre aligned

nikit19, 2017-09-29:
**Actual Behaviour**

Currently nav_header_main is not centre aligned which doesn't look good.

**Expected Behaviour**
nav_header_main should be centre aligned

**Steps to reproduce it**

Open the app and look at the navigation drawer

**Screenshots of the issue**

**Before**
[image]

**After**
[image]

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2017-09-29:
Hi @nikit19 ! Current implementation follows [material guidelines](https://material.io/guidelines/patterns/navigation-drawer.html#)! I don't think this is necessary!

nikit19, 2017-09-29:
@CloudyPadmal  okay I'll close the issue

closed 2017-09-29 by nikit19