fossasia/pslab-app #652: OscilloscopeActivity doesn't rotate on screen rotation

nikit19, 2017-09-30:
**Actual Behaviour**

OscilloscopeActivity remains fixed on screen rotation

**Expected Behaviour**

It should remain in landscape mode but allow the user to rotatate the screen if the user desires to do so

**Steps to reproduce it**

Open the app. Select applications and click on Oscilloscope

**Would you like to work on the issue?**

Yes

akarshan96, 2017-09-30:
@nikit19 we have decided to keep Oscilloscope in Landscape mode only. This ensures maximum area to graphs.
Cheers

nikit19, 2017-09-30:
@akarshan96  yes it should be in landscape mode. But what I meant is it should allow rotation in landscape mode.  What do you think ?

CloudyPadmal, 2017-09-30:
Hi @nikit19 ! That would be fine I guess; rotating only in landscape modes. Send a PR when ready! :+1:

commit 3254d69b (2017-10-02): Fix #652 landscape rotation works now (#653)
closed 2017-10-02 by mariobehling