fossasia/pslab-app #657: Create the Transients RLC Experiment

asitava1998, 2017-10-06:
**Actual Behaviour**
The Transients RLC Experiment has not been created so far

**Expected Behaviour**
Create the Transients RLC Experiment

**Steps to reproduce it**
Main Menu -> Saved Experiments -> Electrical Experiments -> Transients RLC Experiment

**LogCat for the issue**
None

**Screenshots of the issue**
None

**Would you like to work on the issue?**
Yes. In case a Codeheat developer wants to work on it, leave a comment here.

NamanRawat1, 2017-10-06:
Can I work on it?

asitava1998, 2017-10-06:
@NamanRawat1  Yes sure. Do you have the PSLab Device? If not , then I will update this thread with some resources for your help.

NamanRawat1, 2017-10-06:
No I don't have PSLab devices @asitava1998

asitava1998, 2017-10-09:
[image]
[image]
As you can see there are elements like a plotting region, a spinner to adjust the x axis scale, buttons to set the offset and curve fitting.
This experiment is complicated and see if you can manage without the device. Personally I would recommend you to start with something easier like the `RC Integrals, Derivatives Experiment` in issue #659

NamanRawat1, 2017-10-09:
ok i will work on issue #659 first. @asitava1998

CloudyPadmal, 2018-05-28:
Issues related to experiments will be considered once the Android app is functional

closed 2018-05-28 by CloudyPadmal