fossasia/pslab-app #659: Create RC Integrals, Derivatives Experiment

asitava1998, 2017-10-06:
**Actual Behaviour**
The `RC Integrals, Derivatives` Experiment has not been created so far.

**Expected Behaviour**
Create the `RC Integrals, Derivatives` Experiment

**Steps to reproduce it**
`Main Menu` -> `Saved Experiments` -> `Electrical Experiments` -> `RC Integrals, Derivatives`

**LogCat for the issue**
None

**Screenshots of the issue**
None

**Would you like to work on the issue?**

NamanRawat1, 2017-10-23:
Check the code at -https://github.com/NamanRawat1/pslab-android/commit/e199a93887da4d24471dab9109e462865f0d0065

Showing unexpected output-
[image]
@asitava1998

NamanRawat1, 2017-10-24:
Add .md extension and tried with some other file still the same output @asitava1998

asitava1998, 2017-10-28:
@NamanRawat1  I left at a comment in your code for the change to get the code running. Fix it soon.

NamanRawat1, 2017-10-30:
https://github.com/NamanRawat1/pslab-android/commit/81dd0f74bb5f14e2e31b0e1cd19647456908e87a
Change the .md file name still the same output 
@asitava1998

asitava1998, 2017-10-30:
@NamanRawat1  It is not about the filename. The experiment's name is `RC Integrals, Derivatives` , but you wrote it as `RC Integral` . That's why the file isn't rendered correctly.

CloudyPadmal, 2018-05-28:
Fixed in #678

closed 2018-05-28 by CloudyPadmal