fossasia/pslab-app #660: Create Voltage controlled Low Pass Filter Experiment

asitava1998, 2017-10-06:
**Actual Behaviour**
The `Voltage controlled Low Pass Filter` Experiment has not been created so far.

**Expected Behaviour**
Create the `Voltage controlled Low Pass Filter` experiment.

**Steps to reproduce it**
`Main Menu` -> `Saved Experiments` -> `Electrical Experiments` -> `Voltage Controlled Low Pass Filter`

**LogCat for the issue**
None

**Screenshots of the issue**
None

**Would you like to work on the issue?**

divyabaid16, 2018-01-04:
I would like to work on this issue.

CloudyPadmal, 2018-01-06:
@divyabaid16 Sure! Please send us a PR once you are done! 👍

CloudyPadmal, 2018-05-28:
Issues related to experiments will be considered once the Android app is functional

closed 2018-05-28 by CloudyPadmal