fossasia/pslab-app #675: Conversion of speed of sound experiment to Docmd fragment

NamanRawat1, 2017-10-26:
**Actual Behaviour**
The experiment is running using HTML fragment and showing some fluctuations.
Screenshot-
[image]

**Expected Behaviour**
Experiment show in standard format of .md used in all other experiments 

**Steps to reproduce it**
saved experiment->physics experiment->speed of sound

Let us know if this issue should be assigned to you 
Yes

Avjeet, 2018-02-14:
I want to work in this issue can i have more details..

abhinavraj23, 2018-02-21:
I also want to work in this issue

asitava1998, 2018-02-28:
Fixed in PR #733

closed 2018-02-28 by asitava1998