fossasia/pslab-app #679: Remove play button from experiments screen

CloudyPadmal, 2017-11-30:
**Actual Behaviour**

The play button in saved experiments is not useful and it is redundant.

**Expected Behaviour**

The button needs to be removed from the code as well as the layout

**Steps to reproduce it**

1. Open app
2. Go to `Saved Experiments`

**LogCat for the issue**

> Not available

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

This could be a task for GCI '17

srv-twry, 2017-12-07:
@CloudyPadmal can i take it as my first issue ? I'm not a participant of GCI'17 though.

CloudyPadmal, 2017-12-07:
Sure sure!!! Go ahead :) Send us a pull request!

commit f39fbe06 (2017-12-25): Fixes #679 Removed redundant play button from the experiments screen (#681)
closed 2017-12-25 by CloudyPadmal