fossasia/pslab-app #684: Optimization of UI for better response in larger screens

Rimjhim28, 2017-12-23:
**Actual Behaviour**

Currently, the UI for all activities are just streched out for larger screens.

**Expected Behaviour**

A better option would be to optimize the UI to respond better for devices with larger screens and landscape layouts specifically according to material design guidelines such as, an open navigation drawer for tablet view, having more number of items populated on screen when on larger devices, etc.

**Steps to reproduce it**

Redefining of layouts for various screen widths as specified by material design guidelines.

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

[image]

[image]

[image]

[image]

[image]

[image]

The views can be redefined to have a more centrally aligned text view with larger font sizes.

**Would you like to work on the issue?**

YES, I would like towork on the issue

Rimjhim28, 2017-12-23:
@CloudyPadmal @nikit19 Is this issue valid enough to be worked on?

CloudyPadmal, 2017-12-25:
@Rimjhim28 sure! Go ahead and send a PR! 👍

closed 2018-01-05 by Rimjhim28