fossasia/pslab-app #686: Usage of string resources instead of Hardcoded string

Rimjhim28, 2017-12-25:
**Actual Behaviour**

Though some hardcoded strings were removed through PR #674 , there are many other hardcoded strings throughout project that could be replaced with the usage of string resources.

**Expected Behaviour**

Usage of string resources for android:text to include any string

**Steps to reproduce it**

Using string resources to replace the hadcoded string.

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

Screenshot of Lint output to indicate the usage of hardcoded string throughout the project:

[image]

**Would you like to work on the issue?**

YES, I would like to work on the issue.

Rimjhim28, 2017-12-25:
@CloudyPadmal Is this issue valid enough to be worked on?

CloudyPadmal, 2017-12-31:
Sure! Go ahead! :+1:

commit 29d6914c (2018-01-04): Fixes #686 Using string resources instead of hardcoded strings (#689)
closed 2018-01-04 by CloudyPadmal