fossasia/pslab-app #687: Adding full screen views for images in saved experiments

Rimjhim28, 2017-12-30:
**Actual Behaviour**

The various images and screenshots provided in saved experiments are non clickable and may become quite difficult to view on devices with small screen widths

**Expected Behaviour**

A better option would be to make the images clickable such that the user could be directed to full screen view for the images.

**Steps to reproduce it**

Not applicable

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

YES, I would like to work on the issue.

Rimjhim28, 2017-12-31:
@CloudyPadmal @mariobehling Can this be done on the project?

CloudyPadmal, 2017-12-31:
@Rimjhim28 Do you mean when clicking on an image will render it in a different view where we can zoom in and out the image to view it? Please note that the `.md` files are rendered using `br.tiagohm.markdownview` so just see if you can get clickable events!

Rimjhim28, 2017-12-31:
@CloudyPadmal I mean to add a full screen view for images by adding image buttons ro simply buttons with alpha set 0 to make them invisible. Click would direct to a new activity with full view of the image, that is , an activity just with an image view. No, just a full screen view without any zoom in and out feature.
If you assign me here, I will work on this and update you soon.

viveksb007, 2018-02-12:
@Rimjhim28 any updates on it?

Rimjhim28, 2018-04-18:
@viveksb007 @CloudyPadmal sorry for the delay.
Can you please help here on how this should be done?
I think we can add a button with no visibility such that clicking on images would also respond and clicking on buttons and then the user can be directed to a different activity with full screen view of the screenshot.

CloudyPadmal, 2018-04-19:
There's a fix by @ThisIsNSH made in #793 where he zooms the content on the page itself.

CloudyPadmal, 2018-07-16:
Experiments have been transferred to pslab-experiments

closed 2018-07-16 by CloudyPadmal