fossasia/pslab-app #688: Usage if dimens.xml instead to hardcode values

Rimjhim28, 2017-12-31:
**Actual Behaviour**

There is usage of hardcoded values throughtout the xml files of the project for layout widths,heights and margins, text sizes and elevation.
It would be better if those values could be used through dimens.xml to provide a better option for reusability and rectification.

**Expected Behaviour**

Usga of dimensions through dimens.xml.

**Steps to reproduce it**

Not applicable

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Not applicable

**Would you like to work on the issue?**

YES, I would like to be assigned for the issue.

Rimjhim28, 2017-12-31:
@CloudyPadmal Is this issue valid enough to worked on?

CloudyPadmal, 2017-12-31:
Yes please work on! :+1:

closed 2018-01-05 by Rimjhim28