fossasia/pslab-app #692: Inclusion of warning dialogs of Applications

Rimjhim28, 2018-01-05:
**Actual Behaviour**

Currently  the segments defined under applications do not warning dialogs, expect Oscilloscope.
When back button is pressed from oscilloscope, the user if first presented with a warning dialog whether he wants to close oscilloscope or not, to implement uniformity.
A similar warning dialog could be implemented for:
1.Control
2.Logical Analyzer
3.Data Sensor Logger
4.Sensor QuickView
5. Wave Generator

**Expected Behaviour**

Warning dialog appearing in all the Applications

**Steps to reproduce it**

Navigation Menu -> Applications -> Oscilloscope -> Back Button (A warning dialog appears)
Navigation Menu -> Applications -> Control -> Back button (The activity is exited directly without any warning)

**LogCat for the issue**

Not Applicable

**Screenshots of the issue**

The current warning dialog for Oscilloscope:
[image]

I intend to include similar warning dialog for all other applications.

**Would you like to work on the issue?**

YES, I would like to work on this issue.

Rimjhim28, 2018-01-05:
@CloudyPadmal Can this be done?

CloudyPadmal, 2018-01-06:
@Rimjhim28 Yes please go on! 👍

commit 876c3cea (2018-01-14): Fixes #692 Inclusion of warning dialogs when back button is pressed (#693)
closed 2018-01-14 by CloudyPadmal