fossasia/pslab-app #712: Figures missing in many saved experiments

abhinavraj23, 2018-02-10:
**Actual Behaviour**

In many saved experiments it has been mentioned "as shown in the figure but the figure is missing"

**Expected Behaviour**

The figures should appear

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

sharmamohit123, 2018-02-14:
I would like to work on this issue.

sharmamohit123, 2018-02-14:
@abhinavraj23 Is it okay if I use png format image instead of svg format.

abhinavraj23, 2018-02-15:
@sharmamohit123 I am already working in this issue, but we can divide the task you can do for the electronics experiment  and I will be doing it for the electrical experiments. Hope it is okay with you!!!

abhinavraj23, 2018-02-15:
@sharmamohit123 prefer svg format.

CloudyPadmal, 2018-02-15:
Yes please use svg format for images.

Vipul999ujawane, 2018-02-18:
I'm working on the electronics experiments.

PR #720 (2018-02-22): Created figures for electrical experiment
Changes: Added figures for the electrical experiment. Partially fixed #712'

Screenshots for the change: 
[image]
closed 2018-02-22 by CloudyPadmal