fossasia/pslab-app #717: Application UI improvement

Avjeet, 2018-02-17:
**Actual Behaviour**

The icons are too big because of which all the icons can't be shown properly inside the screen

**Expected Behaviour**

the size of the icons should be small 

**Steps to reproduce it**

adapters->application adapter

**LogCat for the issue**

no logcat

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**
Yes
I would like to work on this issue.

harsh-2711, 2018-03-18:
Is this still open? Because the view used is Scroll View and so no need of modifying the card views

Avjeet, 2018-03-18:
actually I m trying to create a grid view as recycler view is not required.I will make a PR for that in a day or two.

Avjeet, 2018-03-18:
Actually it looks very better if all the items view in in one page . I will make a PR for it in a day or two.

cweitat, 2018-07-09:
Please close this issue if there's nothing more to edit

CloudyPadmal, 2018-07-10:
Yes @cweitat ! Now this one is fixed.

closed 2018-07-10 by CloudyPadmal