fossasia/pslab-app #724: bug in oscilloscope activity

Avjeet, 2018-02-20:
**Actual Behaviour**

On selecting the "enable xy plot" there is a null pointer exception 

**Expected Behaviour**

it should show the xy axis in the graph and no null pointer exception 

**Steps to reproduce it**
activity->oscilloscope activity

Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

**LogCat for the issue**

[image]

**Would you like to work on the issue?**
yes

Avjeet, 2018-02-20:
i have resolved the issue mentors plz review my previous PR #713 so that I can submit new one.

viveksb007, 2018-03-08:
Fixed in #728

closed 2018-03-08 by viveksb007