fossasia/pslab-app #731: Travis throwing emulator outdated warning

CloudyPadmal, 2018-02-22:
**Actual Behaviour**

During Travis build, it throws an error saying emulator is outdated!

**Expected Behaviour**

Emulator needs to be run to test unit tests and etc. Outdated prevents it from starting

**Steps to reproduce it**

Check build log for Travis

**LogCat for the issue**

```
Your emulator is out of date, please update by launching Android Studio:
 - Start Android Studio
 - Select menu "Tools > Android > SDK Manager"
 - Click "SDK Tools" tab
 - Check "Android SDK Tools" checkbox
 - Click "OK"
```

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

One of the new contributor can take this issue as they will get in-depth knowledge on the PSLab Travis build system.

Vipul999ujawane, 2018-02-27:
I want to take up this issue

abhinavraj23, 2018-02-27:
The "Android SDK Tools" checkbox is already checked in in my android studio

CloudyPadmal, 2018-06-18:
We are not using emulator in Travis anymore.

closed 2018-06-18 by CloudyPadmal