fossasia/pslab-app #736: Bug in the Control Advanced fragment

Avjeet, 2018-02-25:
**Actual Behaviour**

 The value of all the  edittext in arbitrary waveform generator  is being set to zero on pressing the set button.

**Expected Behaviour**

It should not set the value to zero.

**Steps to reproduce it**

activity->control activity->control advanced fragment

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

Avjeet, 2018-02-25:
@CloudyPadmal @viveksb007 @akarshan96 It is due to the phase editext is throwing number format exception as it is expecting a value in integer and the editext has float,mentors plz guide what value is required by the phase (float or integer ) so that I can make the changes accordingly.

CloudyPadmal, 2018-05-27:
Resolved in #748

closed 2018-05-27 by CloudyPadmal