fossasia/pslab-app #738: Animation not working properly

harsh-2711, 2018-02-27:
**Actual Behaviour**

When going from one page to another, the texts overlap on each other due to bad timings of animation.

**Expected Behaviour**

First, the text of current page should fade and then the text of new page should come up.

**Steps to reproduce it**

1. Go to applications from side menu

2. Switch to Saved Experiments tab from the side menu

**LogCat for the issue**

No LogCat

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

ThisIsNSH, 2018-03-17:
Please close this issue... As it is resolved.

akarshan96, 2018-03-18:
Fixed in #749

closed 2018-03-18 by akarshan96