fossasia/pslab-app #752: Enhance "Report Issue"

ThisIsNSH, 2018-03-08:
**Actual Behaviour**
When Report Issue is clicked then the new issue webpage appears on a web browser not on the app. Due to this the user sees the previous activity which he was on for a second and then the webpage starts loading.

**Expected Behaviour**
Users should be allowed to report issues within the app. 

**Steps to reproduce it**
Remove customTabService in activity->MainActivity.java and add new activity for reporting issues.

**LogCat for the issue**
None

**Video of the issue**
[image]

**Would you like to work on the issue?**
Yes

ThisIsNSH, 2018-03-08:
@CloudyPadmal please assign me this.

CloudyPadmal, 2018-05-28:
Current issue reporting implementation is good enough to handle the task

closed 2018-05-28 by CloudyPadmal