fossasia/pslab-app #754: Shaking Screen when App is auto rotated

Dhiraj240, 2018-03-11:
Actually the app should show a stable view when it is in # portrait mode.
[image]

But when i rotate the phone in # Landscape mode it provides a shaking image
[image]

Expected should be a normal view which was visible in portrait mode and it should not give this awkward shaky image.

Actually this is happening with this ohm's law circuit only.
Hence the steps would be 
**Steps to reproduce it**

Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

There could be a possible changes one could be by creating a new class that extends the # Application and moving initialization code into an overridden onCreate method within the class and by this way the data will not be lost when it rotates because i feel that this may be the cause.

I have just started to work on this project and i think that above variation will eradicate the cause.
**Would you like to work on the issue?**
Yes i can remove this cause.

CloudyPadmal, 2018-03-11:
Hi @Dhiraj240 ! In the **Steps to reproduce it** section, only mention how to recreate the issue.

Dhiraj240, 2018-03-11:
Ok!! There could be a possible changes one could be by creating a new class that extends the  Application and moving initialization code into an overridden onCreate method within the class and by this way the data will not be lost when it rotates because i feel that this may be the cause.

Dhiraj240, 2018-03-11:
please assign this to me.

harsh-2711, 2018-03-12:
Nothing like this happens on my device (One plus 5T Android 8.0) All the functionalities are running as per the requirement

CloudyPadmal, 2018-03-12:
Hi @Dhiraj240 ! Is this issue reproducible? Mine also looks fine. If you think there can be an upgrade, please send us a PR and let's review! :+1:  And also make sure to use the development branch. The screenshot looks like it's from the master branch.

CloudyPadmal, 2018-05-26:
Not reproducible

closed 2018-05-26 by CloudyPadmal