fossasia/pslab-app #755: Details of the Save Circuit getting vanished after pressing utilities Inside Experiment Document

Dhiraj240, 2018-03-11:
**Actual Behaviour**

[image]
[image]
[image]

**Expected Behaviour**

After pressing  experiments when we return back to home it shows **PSlab help file**.
Instead it should revert back the same details after pressing the home.

**Steps to reproduce it**

Home task is getting cleared after pressing utilities and can be restored  by manipulating the Intent as i feel that correct launch behavior has to be created.This happened when someone has put a sorting technique inside the body of Experiment Document.. 

**Would you like to work on the issue?**
Yes

Dhiraj240, 2018-03-11:
@CloudyPadmal  Can i work on this project ?

akarshan96, 2018-03-12:
@Dhiraj240 sure you can

Dhiraj240, 2018-03-12:
Thank you,will update as soon as possible.

CloudyPadmal, 2018-07-16:
Experiments have been transferred to pslab-experiments

closed 2018-07-16 by CloudyPadmal