fossasia/pslab-app #770: Help & feedback layout not having toolbar

bravoo84, 2018-03-25:
**Actual Behavior**
Just white void layout comes up.

**Expected Behavior**
Should have FAQ and help instructions.

**Steps to reproduce it**
Simply implementing toolbar layout and back button functionality on it. 

**Screen-shot of the issue**
[image]

**Would you like to work on the issue?**

_**Yes**_,I would like to work on this.

harsh-2711, 2018-03-25:
Ya.. This part and Design Experiments part both needs to be worked upon

bravoo84, 2018-03-25:
Ok..I will be making the change there also

bravoo84, 2018-03-25:
**fixes issue #770**

**Gif of the fixed issue**
[image]

**Apk of updated app**
[app-debug.zip](https://github.com/fossasia/pslab-android/files/1845671/app-debug.zip)

bravoo84, 2018-03-25:
@CloudyPadmal please review my 2 commits