fossasia/pslab-app #778: Make the schematics and screenshots zoomable

ThisIsNSH, 2018-04-09:
**Actual Behaviour**

Currently the app does not allow user to click on the image (schematics & screenshot) and zoom it.

**Expected Behaviour**

User should be able to zoom the image he wishes to see.

**Steps to reproduce it**

Every markdown is to be changed where the image code is written. There will be a image source but on clicking it, user will be able to see the image in full size ( mainly on an external source i.e. web browser)

**LogCat for the issue**

None

**Screenshots of the issue**

None.

**Would you like to work on the issue?**

Yes

ThisIsNSH, 2018-04-09:
@CloudyPadmal @akarshan96 @viveksb007 please check if work is to done on this issue?

mariobehling, 2018-04-12:
Features should work offline. Image should be zoomable inside the app.

ThisIsNSH, 2018-04-12:
Definitely.. I will work on this and get back.

closed 2018-05-17 by mariobehling