fossasia/pslab-app #804: Change access to private instead of public

Rimjhim28, 2018-04-20:
**Actual Behaviour**

There is a warning "Access can be changed to private"

**Expected Behaviour**

Most of the classes and methods are declared public which gives a warning in Android Studio but instead they can be made package-private. This will help counter two problems -
a) Easier to debug that class or method
b) Would help to reduce the unnecessary Coupling.

**Steps to reproduce it**

Add steps to reproduce bugs or add information on the place where the feature should be implemented. Add links to a sample deployment or code.

**Would you like to work on the issue?**

Yes.

Rimjhim28, 2018-05-08:
@CloudyPadmal I am busy with my GSoC project in Systers org.
If anyone intends to work on this issue, he can go ahead.

CloudyPadmal, 2018-05-08:
Sure @Rimjhim28 ! :+1:

PR #807 (2018-05-10): fix: changing the access to private
Fixes #804 

Changes: Changed the access of the variables to private wherever applicable

APK for testing:
[app-debug.zip](https://github.com/fossasia/pslab-android/files/1936064/app-debug.zip)
commit 7063e437 (2018-05-21): Fixes #804 Changing the access to private (#807)
closed 2018-05-10 by akarshan96