fossasia/pslab-app #805: Create Design Your Experiment Module

Avjeet, 2018-04-21:
**Actual Behaviour**

This module is currently missing from the current app.

**Expected Behaviour**

This module allow the users to customize their own experiment based on their requirements.

**Steps to reproduce it**

->Create new Activity

->Create view pager inside the activity containing fragments for each of the following tabs.

   1.)Load Profile Tab- This fragment contains a list of files in which the previous setup of the experiments 
                                   which are saved.
   
   2.)Configure Experiment Tab - The user can configure the experiment i.e. set the channels for inputs 
                                                    and channels for reading outputs and then can plot these on graph. 
   
   3.)Measurement Tab - This fragment generate the values of the sweep channels in the form of the 
                                       table.

   4.)Plot Tab - This fragment will show a 2D graph, here the app will plot the values of specific channels 
                       in the graph to show a plotted curve.

**Screenshots of the issue**

Some mockups for the design your experiment are below: 

Configure Your Experiment
[image]

Measurement
[image]

**Would you like to work on the issue?**
Yes

Avjeet, 2018-04-21:
@mariobehling the layouts are in accordance with the desktop repo of the PSlab.

CloudyPadmal, 2018-07-16:
Experiments have been moved to pslab-experiments

closed 2018-07-16 by CloudyPadmal