fossasia/pslab-app #827: Adding links for device description and help&support on home screen.

Avjeet, 2018-05-05:
**Actual Behaviour**

Currently home screen is showing only device attached status.

**Expected Behaviour**

Home Screen should also contain links to device description and button for help & support on the home screen .

**Steps to reproduce it**
Fragments->HomeFragment

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

Avjeet, 2018-05-05:
In this link of "What is PS Lab ?" we can provide them with all the information about the device like its features and where can u buy one etc.
Also this will take up extra space on the home screen and make it look better.

CloudyPadmal, 2018-05-05:
Hello @Avjeet ! Instead of this screen, we'll take the current `Applications` part to the home page. So it's one less click~ :)

Avjeet, 2018-05-05:
yeah but then where will we show the usb attach status ?? @CloudyPadmal

CloudyPadmal, 2018-05-05:
@Avjeet I've added a mockup in [Invision](https://projects.invisionapp.com/boards/9A3LPR0RZ6G/) and invited you to the group. There we can finalize UIs before implementing them!

Avjeet, 2018-05-06:
@CloudyPadmal can we implement it like this- show this home fragment only when the device is not connected as an empty activity .
When the device is connected this empty activity will simply  be replaced by the recycler view .

CloudyPadmal, 2018-05-07:
@Avjeet Ah you mean "Device not connected" view when it is not connected and only when connected user can access the features?

Avjeet, 2018-05-07:
Yeah, the PSLAB tools contained in recycler view will only be available when device is connected  . Rest of the app can still be accessed without the device.

CloudyPadmal, 2018-05-08:
Sounds good! :+1:

cweitat, 2018-05-28:
#932

PR #897 (2018-05-31): feat: added web view for device description link and improved UI in Home Fragment
Fixes #827 
Added a "What is PS Lab Device" link which points to a **webview** of pslab website and improved the layout of the Home Fragment by adding new icons.

Screenshots for the change: 
[image]
[image]
closed 2018-05-31 by CloudyPadmal