fossasia/pslab-app #850: Control Main activity should have reset button

Avjeet, 2018-05-10:
**Actual Behaviour**

There is no button to reset the the value and the seek bar to default.

**Expected Behaviour**

There should be reset button so the user don't have to seek the bar manually .
The  extra box at the right side can be replaced by a reset button.

**Screenshots of the issue**

Where-ever possible add a screenshot of the issue.
[image]

**Would you like to work on the issue?**

Yes

Avjeet, 2018-05-10:
I have designed a layout, mentors plz comment @mariobehling @CloudyPadmal @akarshan96 

[image]

CloudyPadmal, 2018-05-28:
This will be implemented in the updated power source UI

closed 2018-05-28 by CloudyPadmal