fossasia/pslab-app #855: Adding units to Control UI

cweitat, 2018-05-11:
**Actual Behaviour**
Currently Wave has no unit indication, new user has no clue whether it is peak to peak or frequency 

**Expected Behaviour**
To display units at input fields

State here what the feature should enable the user to do.

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Is open to public

yatri1609, 2018-05-11:
I would like to work upon this, but can you guide me a bit through it @cweitat ?

cweitat, 2018-05-11:
Wave's unit is in Hz so just adding Hz at the end or outside of textbox for Wave 1 and Wave 2 is sufficient.

CloudyPadmal, 2018-05-11:
Sure. I'll assign you