fossasia/pslab-app #858: Experiment Doc. of Voltage controlled Low Pass filter Experiment

yatri1609, 2018-05-11:
**Actual Behaviour**
The font is not proper and even the image is not linked properly.

**Expected Behaviour**
The font must be proper along with the image file.

**Steps to reproduce it**
1. Open Saved Experiments
2. Open Electrical Experiment
3. Open voltage controlled Low Pass Filter 

**LogCat for the issue**

Not required

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**

Yes.

harsh-2711, 2018-05-11:
It's perfectly fine for me.

[image]

What about others ?

abhinavraj23, 2018-05-11:
@yatri1609 this issue is already resolved by me in the pr #844 . Please pull the latest changes in your system :+1:

closed 2018-05-11 by CloudyPadmal