fossasia/pslab-app #859: Enable scrollability in Exp. Setup of Servo Motors in Landscape mode

yatri1609, 2018-05-11:
**Actual Behaviour**

The buttons are not seen in landscape mode.

**Expected Behaviour**

The buttons must be accessible in both the modes.

**Steps to reproduce it**

1. Open Saved Experiment
2. Goto Add-On Module 
3. Open various utilities
4. Open Servo Motor

**LogCat for the issue**
Not required 

**Screenshots of the issue**
[image]
[image]

**Would you like to work on the issue?**
Yes

harsh-2711, 2018-05-11:
The view is indeed scrollable and is working for me
[image]

CloudyPadmal, 2018-05-11:
@harsh-2711 your screen seems bigger, so the buttons are visible. Mine it's stuck at Servo 4 without showing buttons. @yatri1609 please restrict the layout to stay in portrait mode. I think that will make the fix any easier?

CloudyPadmal, 2018-05-18:
@yatri1609 for each PR, have a different branch. Now it's combined with the earlier PR. Please cherry-pick that commit to a new branch and make a new PR!

CloudyPadmal, 2018-05-28:
Issues related to experiments will be considered once the Android app is functional

closed 2018-05-28 by CloudyPadmal