fossasia/pslab-app #873: Oscilloscope communication errors for pin SI1, SI2

Vikum94, 2018-05-13:
**Actual Behaviour**

The SI1 and SI2 is always a square wave

**Expected Behaviour**

Should be a sine wave which responds to the set frequency value at control activity.

**Steps to reproduce it**

Connect SI1 pin to CH1 and SI2 to CH2. Check the Oscilloscope by enabling CH1 and CH2 at Channel parameters. Change the value of Wave 1 and Wave 2 from Control Main fragment and see. Nothing will change

**LogCat for the issue**

Provide logs for the crash here

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-05-13:
Hi @Vikum94 ! For me I'm getting a Sine wave. By any chance have you selected square wave output instead of sine?

Vikum94, 2018-05-13:
Do you mean from Control Advanced fragment?

CloudyPadmal, 2018-05-13:
Yes!

Vikum94, 2018-05-13:
I am using the sine wave.

closed 2018-06-22 by CloudyPadmal