fossasia/pslab-app #874: Oscilloscope cannot handle simultaneous inputs from SQ1 and MIC

Vikum94, 2018-05-13:
**Actual Behaviour**

When try to plot both SQ1 wave from CH1 and Built in microphone, the oscilloscope plot blinks. Cannot handle both the input at once

**Expected Behaviour**

Both waves should be in plotted in oscilloscope having a smooth refresh rate.  

**Steps to reproduce it**

Connect SQ1 with CH1 and enable it. Use built in microphone in from oscilloscope. Select both to plot and enable disable either one and see the changes.

**LogCat for the issue**

> None

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-05-13:
I see lot of leak warnings are there. Do you experience difficulty in running the app in debug mode as that memory monitoring app freezes the app regularly?

Vikum94, 2018-05-13:
No it does not freeze regularly. May be once after app starts. Then it doesn't happen. Notifications are previous leak warnings. I didn't clear them.

CloudyPadmal, 2018-05-13:
Ok great! :+1:

cweitat, 2018-07-09:
What's the status of this?

cweitat, 2018-07-16:
@harsh-2711 @Avjeet @abhinavraj23 @CloudyPadmal due to unavailability of Vikum, who wants to continue this?

CloudyPadmal, 2018-07-16:
@abhinavraj23 can you take this up?

abhinavraj23, 2018-07-17:
Okay @CloudyPadmal i will work on the issue :+1:

PR #1804 (2019-06-18): Fix #874 audio input working fine simultaneously with other channel
Fixes #874 

**Changes**: User can now see output of Ch1 or Ch2 or Ch3 simultaneously with audio output in oscilloscope 

**Screenshot/s for the changes**: 

[image]

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included i [... cut]
closed 2019-06-18 by CloudyPadmal