fossasia/pslab-app #877: UI of application sections needs improvement

abhinavraj23, 2018-05-15:
**Actual Behaviour**

UI of the application section part is not up to the mark

**Expected Behaviour**

The UI of the application section needs improvement

**Steps to reproduce it**

Change the UI in the code.

**LogCat for the issue**

Not applicable

**Screenshots of the issue**

Not applicable

**Would you like to work on the issue?**

Yes

mariobehling, 2018-05-15:
Provide details and explain what needs to be changed and how, please.

Avjeet, 2018-05-16:
A layout similar to this has to be produced 

[image]

yatri1609, 2018-05-16:
@Avjeet and @abhinavraj23  Can I also contribute to this with both of you if both of you are working together? We can divide the work accordingly!

abhinavraj23, 2018-05-16:
@yatri1609 Already me,  @harsh-2711 and @Avjeet are working in it, and it is almost done.

abhinavraj23, 2018-05-16:
@mariobehling @CloudyPadmal @akarshan96  Please review this layout, I will change the layout colour with PSLab colours :+1:

PR #884 (2018-05-26): feat: Added card view for instruments section
Fixes #877 

Changes: Added Card View in instruments section with 1 column in portrait mode and 2 columns in landscape mode. Also great images are added as background.

Screenshots for the change: 
**Portrait Mode**

[image]

**Landscape Mode**

[image]

APK for testing: 
[app-debug.zip](https://github.com/fossasia/pslab-android/files/2015627/app-debug.zip)
closed 2018-05-26 by CloudyPadmal