fossasia/pslab-app #878: Create the wave generator module

Avjeet, 2018-05-15:
**Actual Behaviour**

Currently on clicking the wave generator activity it is showing just a blank screen

**Expected Behaviour**

On clicking wave generator it should open a activity with functionality to generate sine wave , square wave,triangular wave ,saw tooth and should be able to change its duty cycle , phase , amplitude.

**Steps to reproduce it**
Fragment ->Wavegen Fragment

**Would you like to work on the issue?**

Yes

Avjeet, 2018-05-15:
I have created a Layout for the wavegen activity

[image]

CloudyPadmal, 2018-05-15:
Here we need to add some stuff;
---
### Sine Wave
  - There are two pins and we can set two frequencies to those two with a phase difference which also we can set

### Square Wave
  - Similarly we have 4 pins with 4 different settings

### Saw Tooth Wave
  - This is actually a `Triangular Wave` with similar settings to `Sine Wave`

Avjeet, 2018-05-16:
I have thought of creating a view pager with two tabs :- 
1.)**Wave Generator** - For producing sine/square waves using **w1,w2**
2.)**Configure PWM** - For producing PWM outputs using **sq1,sq2,sq3,sq4**

Here the mockups I have created:-
[image]
[image]

mariobehling, 2018-05-17:
Please provide screenshots of existing devices that offer this functionality. How does the UI look like on these? Please follow the standard views of those devices.

Avjeet, 2018-05-17:
@mariobehling This is the existing Ui of the device in android app providing same UI 
[image]

Avjeet, 2018-05-17:
As WaveForm Generator and Configure Pwm are two different things but both of them produce signals, so I thought to separate them in two tabs but put them inside single "Wave Generator Device Activity" as shown by the layouts above this

mariobehling, 2018-05-18:
> Please provide screenshots of existing devices that offer this functionality. How does the UI look like on these? Please follow the standard views of those devices.

What UI do existing devices on the market use. Provide links and details. How is the UI of hardware and digital devices providing the same functionality? What are professionals using? Provide photos, screenshots etc. and follow industry players and existing designs in the implementation.

Avjeet, 2018-05-18:
@mariobehling ok got it, I will update my UIs according to the Function Generators in the market and the Desktop Repo.

Avjeet, 2018-05-19:
Here are the screenshots of the physical function generator provided by HP
[image]

and this is the screenshot of another app providing the wave generator 
[image]
I am creating a layout similar this with a screen.

Avjeet, 2018-05-20:
I have created a mockup of the screen: -
[image]

Avjeet, 2018-05-20:
It is not required

CloudyPadmal, 2018-05-22:
Hello @Avjeet ! We can enlarge the view by following modifications;

1. We don't provide modulation in wave generator, so `Modulation` part can be removed. There is a separate experiment for modulation. So we can just omit it from this one.
2. Only (Triangular), (Sine) and (Square) waves. No offset or amplitude control from app.
3. Apart from the dial, it would be better to have a two up and down buttons as in the HP version to adjust smaller frequency values.

Do all those make sense?

The colors need to be same as the application theme.

Vikum94, 2018-05-22:
Overall the UI looks good but,
1. Red color text does not suit for the black background I think. In the reference figure, they have a good color scheme. You can adjust them to suit our theme. 
2. The subgroup saying Channel is not actually the channel. Channel is where you read the input. You have to change it into maybe Wave. It is the pin where the wave is generated.

Avjeet, 2018-05-26:
@mariobehling @CloudyPadmal @Vikum94 @akarshan96 I have edited the layout of the wave generator 
[image]

Vikum94, 2018-05-26:
Functionality wise all ok according to previous discussions. The selected buttons are not highlighted well. Would it be possible to use green color to show the selected buttons @CloudyPadmal Can the theme have that color? 
Still we cannot adjust the amplitude using the application. So is it necessary here @CloudyPadmal May be we will need it in future?

Avjeet, 2018-05-26:
@Vikum94  there is no need to show selected button as monitor can handle can handle that.

CloudyPadmal, 2018-07-06:
Wave Generator module is now Implemented

closed 2018-07-06 by CloudyPadmal