fossasia/pslab-app #880: Redesign Oscilloscope UI

Vikum94, 2018-05-17:
**Actual Behaviour**

Oscilloscope layout contains a left pane with buttons.

**Expected Behaviour**

The buttons should separate from each other and have more meaning and interactivity. The selected setting should be highlighted.  

**Steps to reproduce it**

Choose Oscilloscope in Instruments

**LogCat for the issue**

> None

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

mariobehling, 2018-05-17:
Please describe more in detail, what the buttons should do exactly. "More meaning and interactivity" does not really tell us what you plan to show here.

Vikum94, 2018-05-17:
The buttons to select and change different parameters in the oscilloscope are shown in the right pane.

- There is no clear boundary between buttons. 
- When clicked, buttons do not show which one is currently selected.

Vikum94, 2018-05-18:
Proposed UI - When selected, the button highlights it's background.

[image]

PR #887 (2018-05-21): fix: Oscilloscope function button grouping and color changes
Fixes #880 

Changes: The four buttons channel parameters, timebase and trigger, data analysis and xy plot are separated from each other and when selected, there is a color change in the text background  

Screenshots for the change: 

[image]

APK for testing: 

[app-debug.zip](https://github.com/fossasia/pslab-android/files/2019230/app-debug.zip)
closed 2018-05-21 by CloudyPadmal