fossasia/pslab-app #885: Create the multimeter module

abhinavraj23, 2018-05-18:
**Actual Behaviour**

Currently the multimeter is inside the control unit section.

**Expected Behaviour**

It has to be made a separate module itself.

**Steps to reproduce it**

Create a multimeter fragment inside the instrument fragment and implement all it's functionality.

**LogCat for the issue**

Not applicable

**Would you like to work on the issue?**

Yes

CloudyPadmal, 2018-05-18:
@abhinavraj23 this is a good start! :+1: Can you design a mock-up soon and let us review?

abhinavraj23, 2018-05-19:
[image]

@CloudyPadmal @mariobehling @akarshan96 The multimeter UI can be implemented in a similar way like above. Please comment on the above UI and let me know what changes are expected so that I can start coding it.

CloudyPadmal, 2018-05-19:
Nice work @abhinavraj23 ! Please have a look at [Android Material Design Guidelines](https://developer.android.com/guide/topics/ui/look-and-feel/) and add paddings and shaped accordingly while developing.

akarshan96, 2018-05-19:
Looks neat 👍

abhinavraj23, 2018-05-22:
![multimeterui](https://user-images.githubusercontent.com/32960257/40353897-36dc8c5c-5dd0-11e8-9907-3df08fff63b9.png
@CloudyPadmal @akarshan96 What do you all think about this UI
The points represent radio buttons.
We can also add feature to put arrow only at that  point which is selected by the setVisibility feature.
Please comment

akarshan96, 2018-05-22:
How about using https://github.com/BeppiMenozzi/Knob as a knob?

abhinavraj23, 2018-05-22:
Sure @akarshan96 . I wasn't knowing it so I designed it in the way I knew but I will surely try those for the knobs. Thanks for the help :+1: .

abhinavraj23, 2018-05-22:
Thank you @harsh-2711 . I would surely use the infinite knob :+1: and update the layout

abhinavraj23, 2018-05-23:
@CloudyPadmal @akarshan96 I have implemented the knob libraries(open source) and I have also tested it in one of my other app, so shall I start coding  this layout , it will be the same as above, just the circle is replaced by the knob, please review all the other things about the layout and let me know if I should start developing it.

PR #910 (2018-06-04): feat: created multimeter activity
Fixes #885 

Changes: Created the multimeter activity

Screenshots for the change: 
[image]

APK for testing: [Compress the *.apk file into a rar or zip file and upload it here]:The apk will only make sense if the pr #884 is merged as there is not different multimeter section in the previous version of app so there is no button to open the activity.
closed 2018-06-04 by CloudyPadmal