fossasia/pslab-app #889: Format Logic Analyzer Designing

harsh-2711, 2018-05-19:
**Actual Behaviour**

The designing and also the features of logic analyzer are limited

**Expected Behaviour**

It should contain all latest features and should have good designing

**Steps to reproduce it**

Go to Logic Analyzer in Instruments section

**LogCat for the issue**

Not applicable

**Would you like to work on the issue?**

Yes

mariobehling, 2018-05-20:
Thank you. Please add references and links. Please always provide sources. Also provide info on devices that provide the features, e.g. photos of devices and their UI.

harsh-2711, 2018-05-21:
Options seen on the sidebar in the given layout can be removed while execution and can be used during configuration as it is in present app. Also a bottom time slider can be added as proposed in previous layout.

The remote shown at the bottom can be neglected as it was used for data storage that was generated
[image]

Here we can provide four channels to switch
[image]
[image]
[image]
This part is implemented but the cursor doesn't shows data at which it is pointed. 

Here is the reference [link](http://redpitaya.readthedocs.io/en/latest/appsFeatures/apps-featured/logic/logic.html)

Is this layout for android app suitable ?

CloudyPadmal, 2018-05-21:
How can we select which channels to display and which mode (Falling Edge, Rising Edge, Normal Pulse)?

CloudyPadmal, 2018-05-21:
Yes looks fine! Please update with a mockup!

PR #1007 (2018-06-11): feat: Changed the layout of Logic Analyzer
Fixes #889 #952 #911 

Changes: Changed the layout of logic Analyzer fragment showing graph as decided in the issue tracker. Also, the layout was set to Landscape as the portrait mode doesn't suits the current layout

APK for testing: 
[app-debug.zip](https://github.com/fossasia/pslab-android/files/2086940/app-debug.zip)
closed 2018-06-11 by CloudyPadmal