fossasia/pslab-app #893: SHT21 : Temperature and Humidity Meter UI Improvement

yatri1609, 2018-05-21:
**Actual Behaviour**
Right now only the Temperature and Humidity are recorded and plotted on the graph. I would like to add maximum and minimum for the same and also enable units change from Celsius to Fahrenheit and vice versa. 
Including some UI improvements. 

**Expected Behaviour**
[image]
This is the kind of Thermo Hygrometers available in the market. So I would like to develop similar UI kind of UI with buttons for changing a the units and implementing maximum and minimum. 
I would be uploading the new UI soon.

**Steps to reproduce it**
Check layout [file] (https://github.com/fossasia/pslab-android/blob/development/app/src/main/res/layout/sensor_sht21.xml) 

**LogCat for the issue**
Not applicable

**Screenshots of the issue**
[image]
[image]

**Would you like to work on the issue?**
Yes

yatri1609, 2018-05-22:
[image]
[image]

This is what I'm planning to do both for Humidity and Temperature. Please review it and tell me if any add-ons or changes are needed.

Vikum94, 2018-05-22:
You can research more and post some more pictures. There are lot of mobile applications developed to measure temperature and humidity using a sensor. You can adjust your UI accordingly.

[Check this](https://www.amazon.com/Thermometer-Hygrometer-Temperature-Humidity-Smartphone/dp/B019178SY2)

[image]

yatri1609, 2018-05-25:
[image]
 What do you think about this?

CloudyPadmal, 2018-05-31:
@yatri1609 how is the progress? Can we add both humidity and temperature data in the same window without using two tabs just like in the app suggested by @Vikum94 ?

yatri1609, 2018-06-02:
@CloudyPadmal Ya I think we can do that and then down we can show buttons which will open up graphs or we can make the icons clickable.

yatri1609, 2018-06-04:
@CloudyPadmal  Do we need to implement the color as per the the temperature and humidity or just put the icons and a mark on the scale?

CloudyPadmal, 2018-06-04:
Design with the app theme would be great!

cweitat, 2018-07-09:
Can I know why is it difficult to implement? What problems did you faced? If you faced problems displaying 2 data in one graph, do ask any one who did oscilloscope to assist. 
Some graphics like the thermometer and the water droplet sign can be excluded. Both temp and humidity should be in the same chart with different line colours. 
Temperature font size doesn't need to be so big. Have a look at Lux Meter. That font size is good enough. I'm sure you can handle the design.

yatri1609, 2018-07-12:
@cweitat Ohk the difficulty was in implementing the thermometer and the water droplet so ya I guess other than that I would be able to handle it now. I'll submit a mockup for the same soon

cweitat, 2018-08-08:
@yatri1609 what's the progress for this?

yatri1609, 2018-08-08:
I guess Harsh was also implementing the temperature and humidity sensor according to the updated sensor list. I have worked half way through this from my part.

yatri1609, 2018-08-08:
@cweitat I'll submit suggested layout by tomorrow

yatri1609, 2018-08-08:
@cweitat I guess the temperature and humidity sensor is now DHT11 #1172 rather than SHT21 so I'm not working on this.

cweitat, 2018-08-09:
@yatri1609 Are you working on it halfway already? it is okay to implement other sensors also. Doesn't necessarily need to be DHT11. The more sensors the device can work with the better

yatri1609, 2018-08-09:
@cweitat ohk I'll continue with this but it will be better to have a same UI for both as both measures temperature and humidity.

yatri1609, 2018-08-09:
@cweitat I'm almost done with the UI which will be similar to the Lux meter and Barometer with just single graph showing both the data in it.

cweitat, 2018-08-09:
@yatri1609 do you have screenshots? so of course the ideal situation is that the device is able to automatically detect DHT11  or SHT21

yatri1609, 2018-08-09:
[image]
Is this kind of layout fine, with 2 meters showing temperature and humidity both? And then chart to show the graphs? @cweitat @CloudyPadmal

cweitat, 2018-08-09:
Please check out #1349

yatri1609, 2018-08-09:
@cweitat What do you think about this ? I had already suggested earlier 
[image]

cweitat, 2018-08-09:
@yatri1609 Don't need the section boxes. The previous design is good enough. Let's finalise the standard design to use first before proceeding

yatri1609, 2018-08-09:
@cweitat Yes sure, I'm currently stopping my working on this.

CloudyPadmal, 2019-06-27:
There is a UI available with mentioned min max values with value changes

closed 2019-06-27 by CloudyPadmal