fossasia/pslab-app #894: Lack of guidance on how to connect a sensor to PSLab device

Vikum94, 2018-05-21:
**Actual Behaviour**

Existing Sensors instrument does not show the list of sensors that can be used with PSLab. Using the scan button, app will try to find if any sensor is attached or not. 

**Expected Behaviour**

It is better to show the compatible list of sensors in the Sensors activity and when selected, rather than showing a message that sensor is not connected, can include a pop-up screen to show how to connect the sensor to the PSLab. For each and every different sensor, there should be a clear image/gif on steps to connect the sensor to the PSLab.   

**Steps to reproduce it**

 > Not applicable

**LogCat for the issue**

> Not applicable

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes. But open to anyone as there are lot of sensors compatible with PSLab.

cweitat, 2018-05-28:
Few solutions here:

1.  created a list tested by us
2.  let users add to the list directly
3.  let users submit the sensors number to us and we try it

On top of that, the list is best to be searchable by filters (e.g. type of sensors, alphabetical, etc) instead of just a scrollable one

Please discuss which is the best way to implement this

CloudyPadmal, 2018-06-21:
Sensors will be implemented as individual instruments. There will be guides available in each sensor view with corresponding issues.

closed 2018-06-21 by CloudyPadmal