fossasia/pslab-app #902: Consider reducing UI tests

iamareebjamal, 2018-05-22:
There are only 2 UI tests in the app right now. It means that the test coverage in UI tests is small. Hence, the huge disadvantage of about 10 minute build time and waiting for emulator outweighs the advantage of having such small set of tests.

Either UI tests should cover lot of cases (not recommended as unit tests should handle most of this) or UI tests should be disabled and replaced with counter unit tests which will run in milliseconds rather than 10s of minutes

One of the test is acually a dummy test.
Right now navigation of app is tested via UI tests. I would say that this is not such an important test but can be easily replicated in the abstracted unit test. If not, I recommend roboelectric to do the same test on JVM. Although not as fast as a unit test, it will run in 1-2 seconds as compared to current situation.

This can be taken up after crucial bug fixes and features but will significantly improve the build time and hence development iterations

CloudyPadmal, 2018-05-22:
Agreed @iamareebjamal ! :+1: Current UI tests do check only if the starting views are up. This can be tested by other means.

iamareebjamal, 2018-05-27:
@CloudyPadmal Where are we on this front?

iamareebjamal, 2018-05-28:
This is still not resolved

PR #909 (2018-05-27): test: removed Travis build time consuming UI tests
Fixes #902 

Changes: 
- Older UI tests were removed to facilitate faster Travis CI build time.

Screenshots for the change: 
> Not available

APK for testing:
> Not available
PR #938 (2018-05-29): chore: removed device tests from travis
Fixes #902

Changes: 
- Removed device connected tests to improve build time.

Screenshots for the change: 
> Not available

APK for testing: 
> Not applicable
closed 2018-05-29 by CloudyPadmal