fossasia/pslab-app #931: Voltage dropdown pops up when opening oscilloscope

cweitat, 2018-05-28:
**Actual Behaviour**
The voltage dropdown to select CH1 pops up automatically when opening oscilloscope 

**Expected Behaviour**
Dropdown should be kept closed till user press on it

**Steps to reproduce it**
Open oscilloscope and you will see it

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
@abhinavraj23 or @Vikum94

cweitat, 2018-05-28:
#816

cweitat, 2018-05-28:
reference #816

Vikum94, 2018-05-28:
@cweitat @CloudyPadmal This does not happen with either the application in pslab-android/development or in Play store. Can you please check again.

cweitat, 2018-05-28:
https://drive.google.com/open?id=11uOxHarBdiVYPYUkR2eHzxpX7FJ5nLw2
Have a look at the attached video

cweitat, 2018-05-29:
dropdown doesn't appears on Android 8
issue open to more feedback from other users

cweitat, 2018-05-30:
This doesn't happen to another Android 7.0 phone tested. Probably just my phone.
So this issue can either leave it open to or @Vikum94 @CloudyPadmal can close it.

CloudyPadmal, 2018-05-30:
Yea @cweitat it doesn't looks like a reproducible issue. Let's close it then :)

closed 2018-05-30 by CloudyPadmal

cweitat, 2018-06-22:
Saw another device that has this issue today: Samsung S7 Edge